utra-robosoccer/soccerbot #13: Write script for automated installation for the soccerbot onto the robot.

Vuwij, 2018-08-29:
Try docker or something

HPrickettMorgan, 2019-10-14:
@Vuwij We're already building the whole thing within a Docker container on Travis, if you can run me through the manual installation procedure I can work on this issue.

HPrickettMorgan, 2019-10-17:
@Vuwij looking at this further, it seems like the vast majority of the CMakeLists.txt files just aren't set up for installation, what is wrong with just catkin build -> make install on the robot? Compiling on docker is already a pain because the Jetson is on an ARM chip, so we'd need an arm build server/to set up QEMU in docker.

Vuwij, 2019-10-17:
Yeah that's a good idea. You seem pretty motivated. Currently I actually do not know the procedure for installation and we are just rebuilding on the robot. Which isn't too bad based on the build times. I am also imagining using a docker container to push things onto the robot. My previous company had this service called cloud deployment. Which allows you to containerize an application and deploy it to the robots you want. Well discuss more on this in meeting but currently. Our focus is on simulation and making the tests for the different robot movements

HPrickettMorgan, 2019-10-17:
After playing around a bit more, it would be super easy to move all the packages to a "build on CI/CD and pull down binaries on to the robot" except darknet_ros. Darknet wants to link against the specific CUDA version it's going to run on, so it wants to be built on the robot/on a Jetson TX2 build server.

HPrickettMorgan, 2019-10-17:
I am currently working on getting darknet_ros building on CI and then saving those artifacts to speed up our main CI builds. This will also let us generate binaries for the non-darknet packages and pull those down instead of building on-robot. I'm currently guessing at what needs to get installed+where, I anticipate a fair bit of troubleshooting whenever we get to actually loading this on a robot so once it's close I can put it on the back burner until we're out of crunch time.

Vuwij, 2019-10-17:
Alright sounds like a plan

closed 2019-12-03 by Vuwij