utra-robosoccer/soccerbot #218: Automate Fusion to URDF to Proto conversion

lkunelk, 2021-05-08:


sabrinalokman, 2021-05-08:
- Fusion2Urdf and urdf2proto script is working
- Collision box added with urdf2proto script 
- Just need to specify joints limit and add sensors on proto file after.
- Useful for next version of robot

closed 2022-03-06 by Vuwij

acrylic-origami, 2022-06-25:
### Improved Fusion to URDF repo

For Fusion to URDF conversion use [**this fork**](https://github.com/SpaceMaster85/fusion2urdf) of the linked fusion2urdf repo. Big features:

- **Allows linked components** so you no longer have to break links
- Does not change the file anymore, so you can continue working without reopening
- Claims to allow nested components (I have not tested this thoroughly)

Minor caveats (possibly of the original repo too):
- Joint and link orders are random
- You cannot suppress features in the timeline