utra-robosoccer/soccerbot #453: Leg Assembly Parts with MX64 Parts

Vuwij, 2022-04-07:


Shah-Rajab, 2022-05-16:
Current planning:
- 2 x xl430 for hip joints that rotate along the z-axis. Buy 1-2 more for spares.
- 4 x xm430 with 1-2 spares
- 6 x MX64 with 2 spares
@manx52 to update the team with our latest inventory of Dynamixel servos.

Location of the MX64s to be decided.

manx52, 2022-05-17:
So it seems like for legs:
xl430: leg_motor_0
xm430: leg_motor_1, leg_motor_2
mx64: leg_motor_3, leg_motor_4, leg_motor_5

Need to verify with velocity data

Shah-Rajab, 2022-05-28:
From @Vuwij :
[image]

Shah-Rajab, 2022-05-28:
@acrylic-origami is the order looking good?

Shah-Rajab, 2022-06-04:
From the pic:
1: MX28
2,3: xm430
4,5,6: MX64

Shah-Rajab, 2022-06-04:
@mx266

closed 2022-06-21 by manx52